prusa3d/MM-control-2.0 #1: kiCAD files do not reflect actual MMU2 production board.

CacikeOtoao, 2018-09-23:
@ok1hra Production boards of the MMU2 that people are getting are showing re work to the vias (to the upper right of the sheep logo) after manufacturing that are not reflected on the Source files provided here. Keep in mind that these files were made public after the MMU2 was released and corrections as described above had been made to the boards. Please provide updated documentation or comments regarding these changes as good Open Source practice. Providing outdated information to the community is not a good practice since the community depends on these files to replicate the boards and Prusa Research preaches about Open Source.

ericclinedinst, 2018-09-23:
My level of surprise is absolutely zero.

ok1hra, 2018-09-23:
@CacikeOtoao thanks for comment. Can you show me this difference? Because revision 0.3 is last production version

ericclinedinst, 2018-09-23:
@ok1hra  for starters R42, d7, r44 are removed on the boards you're shipping.   Where as the BOM has them in place.

ericclinedinst, 2018-09-23:
In some photos it appears that traces have been cut above the sheep on the right side just above his butt.  

Also when will we be getting the bootloader that was spoken about in anot her github thead? You guys said a week ago it would be posted in "days" @ok1hra

ok1hra, 2018-09-23:
And you can post this photos? Without another argument you claim is false.

I am not manage bootloader, I cant help you. Ask on support desk.

closed 2018-09-23 by ok1hra