prusa3d/MM-control-2.0 #2: The orentation of the sockets should be turned by 180°

KarlZeilhofer, 2018-10-10:
Is there any chance, that the sockets will be turned by 180°?  
With the current case of the MMU 2.0 is nearly impossible to unplug the cables when everything is assembled.

ok1hra, 2018-10-11:
We do not have the same feeling.

KarlZeilhofer, 2018-10-18:
No, why not? I think to remember, that the trouble shooting guide says to use an allen key to unlock the securing pin of the communication plug.

domesticatedviking, 2019-09-16:
I very much agree.  When using an enclosure, all of the connections to the MMU2 are in a location that can neither be seen nor reached from the front of the printer, and they all require tools to remove.   The result is that it is VERY difficult to remove the printer for maintenance.   This is why I designed the [underslung control panel mod](https://www.thingiverse.com/thing:3211711).