prusa3d/MM-control-2.0 #10: noise in idle, homing problems, led flickering

SNDR0, 2019-05-05:
like said from @chriswal, adding  a ceramic capacitor 104 between pin 16 and ground of every 74hc595 you solve the problem. (issue known when you are using 74hc595)

darBis, 2019-06-13:
I had recently solved these issues by adding 100 nF capacitors to 74hc595 ICs also.(https://github.com/prusa3d/MM-control-01/issues/124#issuecomment-501355623)
Please add bypass capacitors to those ICs in next revision of the board.
Information about this issue and working solution is relatively hard to find, maybe some list of known issues on wiki would be great.

darBis, 2019-08-24:
This is a serious reliability issue. Since fixing my board by adding those caps i have no more issues with the MMU regarding homing errors and erratic behaviour.

chriswal, 2019-08-24:
i also made a firmware workaround for people who cant modify the board on there own. there is a pull request. or you can get it from my fork

bbeller123, 2020-07-30:
Prusa should add a screen with graphic interface for multi material. I mean it's multi color ?