prusa3d/MM-control-2.0 #12: C60 can cause timing issues

NielschB, 2020-02-20:
I have experienced some issues with my MMU2 board, where all LEDs started flashing and could not get the motors working anymore. All pinned down to capacitor C60 which is connected to the latch input of the shift registers. This caused a slow ramp of the latch signal which resulted in timing issues on the shiftregister side. After reviewing the schematics I wonder why C60 is there in the first place. If its function is to implement a low-pass filter, then a series resistor should be added, now its determined by the Atmega output drive strength.
All together this is not a nice way to do this since a slow signal ramp is applied to a digital input (cross currents etc.) Another thing is that continuously charging and discharging this cap by the MCU is a waste of current.

chriswal, 2020-07-30:
and you resolved this issue by removing C60?

NielschB, 2020-07-30:
Yes, just remove it

NielschB, 2020-07-30:
This C60 might just be there to give non-prusa sellers a hard time ;) I have the feeling that the original Prusa boards might not have this cap assembled.

chriswal, 2020-07-30:
You are right , there should be a resistor in series. but i dont see where there is a timining issue. the enable signals are always on, only the dir signals can change because of the direction the motor is moving. the leds are working also without a problem.
so what is the timing issue you are refering about?

NielschB, 2020-07-30:
Most boards don't show the issue, I have had 2 boards that did have this timing issue. Most probably its related to the output drive strength of the MCU where the cap is connect to, which determines the slope of the latch signal. It might vary over batches. Again, in the end a design should never have an analog LPF on a digital input like this.

When the MCU is shifting out data to the shift registers, the latch signal should toggle at a certain moment in time to load the data in the registers. When the latch signal is being delayed (By the cap and finite drive strength of MCU) the latch signal is crossing the shift register's threshold when next data is already shifted in! which causes corruption of the data in the shift registers.

chriswal, 2020-07-30:
So you had flickering leds and loud motor noises on this boards?

NielschB, 2020-07-30:
Flickering leds yes, motors would just not turn in my case, but I expect anything can hapen.
May I ask why you are interested? Are you experiencing the same issues?

chriswal, 2020-07-30:
i edited the message before.  yes i had this issue and solved it a different way.

chriswal, 2020-07-30:
if the motor doesnt turn then this could be some different problem than mine

NielschB, 2020-07-30:
I did try the buffer capacitors, did not help for me. Then I traced it down to this issue which was confirmed with scope measurements. 
I am wondering if the people who fix their issue with adding decoupling caps to the shift registers are actually suffering this C60 issue and the error is masked with adding buffer caps. Charging / discharging C60 takes quite some current which could explain why adding buffer caps could help. 

Shift registers producing random data due to lack of decoupling caps would mean that the supply drops significantly, which I personally have not observed. (I did measure it actually)

chriswal, 2020-07-30:
but that is a small capacitor and it is charged from the mcu... so for the layout that is further away from the register.
i think to measure this you have to have a very fast scope. is it possible that on these board you had wrong capacitors assembled?
i caan repoduce it on my board if i remove the buffer capacitors.. i also made a workaround in the firmware where you dont change leds while moving the motor .. 
i can try to remove the capacitor..i think they added it because they had problems before with erratic output

chriswal, 2020-07-30:
i didnt build this system ..i had this board ..and some of the people are not able to solder.. so i made this workaround ..
i can solder so i can fix something..i have a 100mhz scope but not around at the moment. and this was not an assumption  
[image]
[image]
from the datashhet

chriswal, 2020-07-30:
but there can be multiple problems leading to the same outcome. thats why i am interested to find the real problem.
there is also a problem with the design of the 5v power on this board.