prusa3d/MM-control-2.0 #15: Design error on +5V net

HenrikBork, 2021-03-28:
Hi,
There seems to be a bug in the design regarding the +5V net.
On my MMU2 i measured 4.2V on the +5V, which is too little for the MCU to run 16MHz. This is seemed to be the reason why my MMU was almost unusable as the MCU resets all the time.
Connecting the usb to a host helped a little on stabillity (and the +5V voltage rose 50-100mV)
I have shorted D1 on the MMU PCB, as a "proof-of-problem", and that raised the voltage to 4.8V and removed all MMU misbehavings. (without the usb connected). In the same bug chasing i also added 1000uF between the shorted D1 and GND. I think C59 was probably just fine as it is. (i was tiered of bug chasing the MMU)
**Please note: this is not a proposal to a solution, just "proof" of some kind of an issue with the +5V (at least on my MMU). I have no understanding of the diode's function in the entire printers power nets.** -although this hack has been working for some time now.

So it seems as if the designer was aiming for generating +5V by means of the TMC2130. But forgot to connect the VSA to >6V. But then again why differ so much from the suggested circuit and component values?...
The solution of using the printers +5V through a diode and a inductor yields too low a voltage.
...so it seems as the +5V was forgotten, but it somehow sometimes almost works...

My hope is that Prusa fix this problem in the hardware design. (and sends me a fixed PCB free of charge ;-)

### You're welcome,
### Henrik B. S.

My setup:
Original Prusa I3MK3 (2-3 years old), since then upgraded with factory parts.
MMU2 bought from Prusa around august 2020.
MMU PCB reads "2019-09 rev 0.4.4"
My ugly hack:
[image]

extra info:
I also reported/discussed this on [https://forum.prusaprinters.org/forum/original-prusa-i3-mmu2s-mmu2-hardware-firmware-and-software-help/i-found-serious-design-error-in-mmu2-electronics-hardware/#post-405106](https://forum.prusaprinters.org/forum/original-prusa-i3-mmu2s-mmu2-hardware-firmware-and-software-help/i-found-serious-design-error-in-mmu2-electronics-hardware/#post-405106)
...My daytime job is to do this kind of work as consultant - so next time just hire me :-)

Chrismettal, 2021-05-22:
Can confirm, my board started to reset every time it needed to unload filament, getting stuck and requiring user input on every single filament change.
Bypassing the diode raised the 5 V rail from 4.6 V to 4.95 V in idle, fixing the resetting issue for now. Before I could see dips under 4.5 V on the rail on filament changes at which point the MCU sometimes reset. 

The capacitor does not seem to be required for me, but I have also added the missing decoupling caps on the shift registers.

Thank you for your find and I sincerely hope Prusa did not abandon the MMU2 and are going to release a new Version of the board (Or maybe at least publish the sources of the most recent version since this repo is behind the actually released boards)

ANTALIFE, 2022-01-03:
Good catch! Looks like quite a big HW bug...
[image]

ChrisPetit, 2022-06-20:
it has been a year...is this being addressed? I haven't had a correct functioning MMU2S since it was first released...

Kugelfang666, 2022-06-20:
Same problem with me, adding a 5V USB PSU to the MMU reduced the frequency of this problem coming up but it still happens every now and then

lemmingDev, 2022-08-27:
So, is there a verified fix for this hardware issue yet?

Kugelfang666, 2022-08-27:
Yep, adding a stepsons to provide the 5V line of the 12 line permanently fixed the issue for me. Check out the tread in the Prusa Forum. All the details are there an I can verify that this is working. Did a multicolor print yesterday and not a single problem!

vladimir-aubrecht, 2022-09-07:
@ok1hra Do you think you could help with some official statement on this issue?

I think any answer would be better than none as it's keeping people in a void of not knowing if it make sense to wait for some kind of official fix (or even new version of the board) or they should just try to fix it themselves/sell mmu2s/...

Unfortunately I started to hit this issue too 😕

JakoobCZ, 2022-10-04:
Hi,
sorry for the late response. We are working on updates to the MMU2S. Our devs are finalizing new smarter firmware, which will bring new features along with improved performance and reliability to the MMU2S - all on the current motherboard. The new firmware is being tested on multiple printers and so far the results look very promising - we're focusing on the elimination of restarts caused by low voltage, reducing required user interventions and overall better user experience. I can't promise any ETA, but we are doing our best to have it done as soon as possible.

Chrismettal, 2022-10-05:
This is great news and I am extremely grateful for an update. Thank you!
A little communication goes a long way, and sharing any progress helps users keep faith in the project.

Can you share the reason why development isn't done in a publically visible branch? Even having the actually released board revision visible here would be an upgrade, as the repo seems to be lagging behind the actually available product version.

GRIFFCOMMca, 2023-06-03:
I have arrived here as the 2019 MK3s with the MMU2s seems to have the same issue, many resets, this only really started to happen recently, we have not modified the printer, we have a Pi Zero, which was upgraded to a Pi Zero 2 in the past 4 months, we found it abit better when moving the data cables for the front LCD away from the data cable from the MMU2s, still has MANY issues with constant resets resulting in 5 green / orange flashing LEDs,

How is the firmware coming as this is really now unusable.

dtryba, 2023-07-30:
> How is the firmware coming as this is really now unusable.

The new firmware is out and the results is that it worsened this bug.

I had this issue when I installed a switch to the MMU2S to easily disable it, so remove the switch again.
The MMU2S worked fairly well with older stock, it worked fine with The Zero Beast firmware. With 3.0.13 and MMU3_3.0.0+814.hex the MMU2S became unworkable. Power issues resulting in "MMU MCU ERROR 04306" and ""MMU SELFTEST FAILED 04325", followed by "IDLER CANNOT HOME 04125" when it got past the first 2. 

Shorting the diode didn't resolve these issues. So that finally drove me to install a DC-DC boost converter between the printer and pin 1. Ramping up the voltage after the diode to 4.9V appears to have given me a workable MMU2S again.

Version of MMU board: 0.3 producion date 2018-04-2x.

JanMrlth, 2023-08-05:
I updated the MK3s+ to 3.13.0-6873 and the MMU2S to 3.0.0 with the result, that I get a 'MMU MCU ERROR 04306' all the time.

puckpuck, 2023-09-09:
I suspect a power problem with my MMU2S after over 2 years of hassle-free usage. Waiting for the new firmware, I finally had time to try it out, and upon first boot, I get the 04306 error.

So, this feels like a hardware issue now. Is Prusa able to help here? Is my only option to buy another MMU mainboard and wait for it to ultimately fail on me again?

SixOThree, 2023-09-29:
Before finding information suggesting a design issue I replaced that MMU mainboard (at my own expense). Then I replaced all of the stepper motors, one by one, again at my own expense. I worked with tech support and posted on and created multiple threads with no suggestion there is a mainboard design issue. Not particularly please with what I am seeing.

jfestrada, 2023-10-02:
Last week I asked to the support about the possibility to buy the PD-Board, they told to me that the PD-Board was only available for the upgrade kits (100€) or new MMU3 purchases, so I'm very angry as a customer because they sold a faulty design and now @josefprusa refuse to sell (not even give as warranty) to the original mmu2s owner the possible solution. So, first will be the new customers and later maybe the old customers. As a customers we should do more noise on social networks because this is a very bad behavior from @josefprusa and @prusa3d with us.

Anyway recently I released an MMU3 version for 10 colors and I found that I could get rid of some MCU errors related with the input current if I change the range for the critical error, of course is not a solution but make the MMU2S with the 3.0.1 firmware more usable, here is the repository [CM3D-Firmware-MMU](https://github.com/ComunidadMaker3D/CM3D-Firmware-MMU)
This is what I changed from 274 to 300. 
[image]

jfestrada, 2023-10-04:
> Hi, If the new firmware 3.0.0 (or newer) is causing issues, please get in touch with our support team to discuss your case and troubleshoot the problem (fastest way: livechat at [prusa3d.com](http://prusa3d.com/)). We can then provide the necessary parts - even a replacement main MMU board if needed and/or a free voucher for the MMU2S->MMU3 upgrade kit, which includes the PD board addon and cable. [Check our Prusa blog article](https://blog.prusa3d.com/original-prusa-mmu3-is-here-multi-material-printing-with-improvements-all-around-now-shipping_80485/) for more information.
> 
> Please let me know if you managed to get things sorted out 🙂

Hi @JakoobCZ , I have reported my issue this morning to the support chat, I'm currently waiting for the answer of the developers. Again, this is a faulty design. So let see what the developers say and which solution offer the support team.

jfestrada, 2023-10-04:
> I updated the MK3s+ to 3.13.0-6873 and the MMU2S to 3.0.0 with the result, that I get a 'MMU MCU ERROR 04306' all the time.

The way to avoid that loop is to modify the value of the VDCACCThreshold variable on the config.h file, I have tested with 300 which will allow 3.8V as minimum. Then build the firmware with utils/build.py script and flash your own version.

SixOThree, 2023-11-03:
Hi All. I have been hesitant to try the suggestions of altering the board. But I did order the MMU3 upgrade kit. And when it arrived I noticed it had a PD-board included. I decided to just give it a shot without using any other parts from the upgrade. After installing the PD-board my printer immediately started working. 

The mainboard is dangling out the back of the unit. But I can move the selector. I can load filament. I can switch to another filament slot and load a second filament there. Heck, I could probably load 5 filaments! I could probably run a multi-color print! 

I have changed no other parts than adding the pd-board.

During the last few years, I have spent countless hours debugging this MMU2 unit. I bought the MMU2S upgrade. I replaced the mainboard. I replaced all of the servos one by one. I replaced the cables (all before finding these threads mind you). 

The most I ever got out of this printer was 5 blinking lights. I considered myself lucky if I could even get a single color print to start. Never mind that they all ended with errors unloading the filament. I had maybe a 1 in 10 chance of resetting the unit and getting a filament to load. Just enough to fight the machine to get a print started. And I refused to convert the printer back to mono filament.

The frustration related to this makes me feel like prusa is not operating in good faith. And seeing this thing working with a little power board makes it all the worse. What I bought was not ever a working product. Replacing every part still resulted in a non-functional product. This MMU2 to MMU3 upgrade should have been free. That's all there is to it. 

This ordeal has 100% ruined the experience of using this printer and looking back, it really has destroyed the hobby for me.

GRIFFCOMMca, 2023-11-03:
Your move PRUSA, its easy, everyone even told you what to do...

jfestrada, 2023-11-03:
Definetly I agree with @SixOThree, @josefprusa sell to us a faulty design, and now they want to avoid the fix just selling it as an upgrade. I asked to the support three times, the last one I made the reference to @JakoobCZ and I have had no answer since then, they say to me that they will reach me with an answer but that never happen. So @josefprusa took from me almost 300€ for a faulty design, now years later they had say "ouch the design has an error" and instead of send the PD-Board to all the customers for free, @josefprusa pretends that we spend another 100€ on the upgrade kit. NO WAY, that's not going to happen, NEVER. I prefer to buy a Bambulab printer than another Prusa machine. Both are close source on electronics and pieces so where is the difference now? I feel very upset with this situation, and by now I will implement this solution --> https://www.printables.com/model/620818-mmu2-5-volt-power-fix?fbclid=IwAR2o2KFAJloaAzRCLWM-nIziLRyKihPp9WNVWtz4ps1AV6q5oigmXqDxc-s#preview which is easy and I can test it right now. So at least that @josefprusa change their mind about the PD-Board I will not recommend a Prusa machine anymore to anybody.

jfestrada, 2023-11-03:
It works. Now the firmware is not raising the error about the lack of voltage.

[image]

jfestrada, 2023-11-03:
> It works. Now the firmware is not raising the error about the lack of voltage.
> 
> [image]

@josefprusa do you belive that we as customers deserve this kind of DIY solutions?

negroISO, 2023-11-16:
Holly hell. One of the reasons I dumped my MK3S+ and MMU2S was this issue! It didn't happen at first but after a while, it just would reset all the damn time. Wouldn't feed filament, wouldn't retract it, just Bloop blink and reset. I need to check my board to see what's up.

phat3r, 2023-12-03:
Chris Riley found an easier way to get 5V ... move one pin ... and a 2 cent du-pont plug ... this has been going on for how long ... https://youtu.be/JAWqIH_U66s?si=WHZxiUF6VFchaMWd&t=1141

puckpuck, 2023-12-05:
> Chris Riley found an easier way to get 5V ... move one pin ... and a 2 cent du-pont plug ... this has been going on for how long ... https://youtu.be/JAWqIH_U66s?si=WHZxiUF6VFchaMWd&t=1141

I just did this... instead of finding a small connector, I used vinyl tape. It works like a charm, and I'm MMU printing right now after nearly 10 months of finicking with it and no MMU in operation.

timolow, 2023-12-11:
@JakoobCZ I contacted support today after my MMU refused to function anymore, got the run around, unable to sell me the PD board without the upgrade kit and said I have to pay full price for an "upgrade kit", As a person who bought the MMU1, MMU2, MMU2S this feels like a cash grab each time I have to pay money to make a product I already paid for work right.

lemmingDev, 2023-12-12:
Lucky AliExpress has you covered ;)
https://www.aliexpress.com/item/1005006221728533.html

phat3r, 2023-12-12:
$40 for part that isn't needed - value ... Just 5 minutes work to de-pin and move the blue wire over to the SPI as per Chris Riley suggestion above ... I have done it on both my MMU2's ... one a clone and one original prusa and has resolved all the power / logic nonsense.   If you want to make it super near ... use a 2c dupont connector - the pin will go into it.

lemmingDev, 2023-12-12:
Yeah - I personally used buck converters for my 3 when it was orinally suggested, though if I were to do it again, I would do as above, and not spend any money...

eoyilmaz, 2023-12-28:
> It works. Now the firmware is not raising the error about the lack of voltage. [...]

Hey, isn't this the 10 color version :)

roster210, 2024-04-05:
This is outrageous we all had faith with them as a company and they have known about this issue and not addressed it for years and the fact the fix is so easy is out of this world, Josef Prusa obviously only cares about more money, they have enough resources to cover this issue, the profit margins on this non working device are out of this world and they still push the issue so  far and do not address it or release a fix until after most peoples or all peoples warrantys are void. I own two and have for years and they are decent printers but that is not a decent company, they have multiple firmware versions that have fried boards then dont honor that fix unless you are in your 1 year warranty range and then they have this.  they have their propritary einsy rambo that is just a pain to  deal with compared to ulti's or even any knock off chinese ones. in hindsight it feels obvious they only care about themselves and Josef, mostly josef and it feels obvious because of the multiple pictures of his arrogant face and the namesake the company. I am not surprised that they evade the issue by ignoring knowledge of it, they seem to b e better at it then the chinese manufactured ones- who have replaced components just off proof of sales, its outrageous this overpriced machine does not give the support you would assume you should get for a $1100 setup at the time. We spend 1000 and prusa cant give us something that costs them <$5 to be able to use our machine. I am going to make as much noise as i canabout this online, its outrageous that this is all it needed and they held onto it until mmu3... on the positive side- i now know a lot more about printers because of fixing and trying to fix this and other issues. Ill never buy a prusa from prusa again and ill  never buy one unless its 1/4 of their price, ill never give that man or his company money again they do not deserve it.

lemmingDev, 2024-04-06:
Yeah - I feel your pain 
For my school, I suggested we buy 7 x MK3S and 3 x MMU2S a number of years ago
We were never able to get the MMU2S working at all and just removed them

Over the years I tried many different 3D parts from DIYers to fix the issues, but it hasn't been until recently when the DIY power fix was worked out that they've now been able to be actually used semi-reliably

I'm in charge of purchasing and we won't be buying Prusa when they're due to be replaced do to this fiasco

The MK3S machines we bought have been fairly reliable and I've been able to maintain them
A Y stepper on an Einsy died a year ago and I replaced it with a clone Einsy off AliExpress
An MMU board died too, and I replaced it with one off AliExpress too
A few heater cartridges, thermisters and fans have also had to be replaced

closed 2024-05-07 by JakoobCZ